ukBaz/python-bluezero #362: Pairing failure for ble_uart.py

vovagorodok, 2022-02-23:
When I'm trying to connect using phone or laptop pairing started and after a while disconnects.
`sudo btmon` shows:
```
...
@ MGMT Event: User Confirmation Request (0x000f) plen 12              {0x0001} [hci0] 30.144586
        BR/EDR Address: 88:83:22:18:06:96 (Samsung Electronics Co.,Ltd)
        Confirm hint: 0x01
        Value: 0x00072264
= bluetoothd: No agent available for request type 2                                   30.146492
= bluetoothd: device_confirm_passkey: Operation not permitted                  [hci0] 30.146542
@ MGMT Command: User Confirmation Negative Reply (0x001d) plen 7      {0x0001} [hci0] 30.145041
        BR/EDR Address: 88:83:22:18:06:96 (Samsung Electronics Co.,Ltd)
< HCI Command: User Confirmation Request Neg Reply (0x01|0x002d) plen 6    #32 [hci0] 30.145102
        Address: 88:83:22:18:06:96 (Samsung Electronics Co.,Ltd)
> HCI Event: Command Complete (0x0e) plen 10                               #33 [hci0] 30.145845
      User Confirmation Request Neg Reply (0x01|0x002d) ncmd 1
        Status: Success (0x00)
        Address: 88:83:22:18:06:96 (Samsung Electronics Co.,Ltd)
@ MGMT Event: Command Complete (0x0001) plen 10                       {0x0001} [hci0] 30.145898
      User Confirmation Negative Reply (0x001d) plen 7
        Status: Success (0x00)
        BR/EDR Address: 88:83:22:18:06:96 (Samsung Electronics Co.,Ltd)
> HCI Event: Simple Pairing Complete (0x36) plen 7                         #34 [hci0] 30.259443
        Status: Authentication Failure (0x05)
        Address: 88:83:22:18:06:96 (Samsung Electronics Co.,Ltd)
@ MGMT Event: Authentication Failed (0x0011) plen 8                   {0x0002} [hci0] 30.259517
        BR/EDR Address: 88:83:22:18:06:96 (Samsung Electronics Co.,Ltd)
        Status: Authentication Failed (0x05)
@ MGMT Event: Authentication Failed (0x0011) plen 8                   {0x0001} [hci0] 30.259517
        BR/EDR Address: 88:83:22:18:06:96 (Samsung Electronics Co.,Ltd)
        Status: Authentication Failed (0x05)
> HCI Event: Disconnect Complete (0x05) plen 4                             #35 [hci0] 30.414942
        Status: Success (0x00)
        Handle: 11
        Reason: Remote User Terminated Connection (0x13)
...
```

ukBaz, 2022-02-23:
With the `ble_uart.py` example it is setup that you should not need to do pairing. Are you using a generic Bluetooth Low Energy scanning and exploration tool?

On a phone I tend to use [nRF Connect](https://www.nordicsemi.com/Products/Development-tools/nRF-Connect-for-mobile). On a windows PC, I use the chrome browser with `chrome://bluetooth-internals`

vovagorodok, 2022-02-23:
the  same for `cpu_temperature.py`?

ukBaz, 2022-02-23:
Yes

vovagorodok, 2022-02-23:
It means that for LE devices, each android application should have implementation of connecting functionality instead of pairing integrated in OS?

ukBaz, 2022-02-23:
I think it is adding user `pi` to group `bluetooth` that is the key bit.
```
pi@raspberrypi4b:~ $ sudo usermod -G bluetooth -a pi  
```

ukBaz, 2022-02-23:
I am not sure I understand your question.

How are you trying to connect to these devices from your Android phone?

vovagorodok, 2022-02-23:
I'm c++/python embedded developer. It's my first experience with bluetooth and android.
 
The goal is to add bluetooth functionality for https://github.com/lichess-org/lichobile. In order to connect physical chess boards to application. The best typescript library that i found for that is `bluetooth-le`: https://github.com/capacitor-community/bluetooth-le#usage.
Will be nice to integrate this functionality simple as possible. Now in setting I implement only boolean option, and I'm not sure if it enough.
[image]
[image]
Should be here some device search and connect functionality?

ukBaz, 2022-02-23:
There is a lot to unpick with those questions and beyond the scope of this library....
As some general points; The first thing to think about is if your situation needs the security of pairing.
Your app will need to know about what it is going to connect to. This can be by knowing the address, name, or service it is advertising.
Trying to develop both ends of the Bluetooth link is typically quite a tricky thing to do especially if it is your first project with BLE. Doing the peripheral first while testing with nRF Connect probably the way to go in most situations. Then develop the central (app) after that.

I've not done anything with TypeScript. My only experience is a talk from https://github.com/thegecko at our local node meetup.

Can I assume that you have the two examples in Bluezero working and I can close this issue?

vovagorodok, 2022-02-23:
Let's close this issue. Thanks for help. For me it's misleading that there no parring. And now I understand that for LE there no parring because each app should have own filter in order to find proper service.

closed 2022-02-23 by vovagorodok