ukBaz/python-bluezero #363: bluetooth on data received

Svein-Tore, 2022-03-04:
Hi 

I see it is possible to send a value to themicrobit with the command   

ubit.text

I just wonder, if I add a bluetooth on data reeceived command in makecode, will it be possible to get that value on the microbit as variable?

ukBaz, 2022-03-04:
The short answer is no. `ubit.text` command is writing to the `LED SERVICE` - `LED Text` characteristic where as the on data reeceived command in makecode is working on the `UART SERVICE` - `TX Characteristic`. More details in:

https://lancaster-university.github.io/microbit-docs/resources/bluetooth/bluetooth_profile.html

The `ubit.uart` command is probably what you are looking for. There is an example of its use in:
https://github.com/ukBaz/python-bluezero/blob/main/examples/microbit_uart.py

If you don't want to receive messages from the UART then you do not need to do the `ubit.run_async()` functionality. So the https://github.com/ukBaz/python-bluezero/blob/main/examples/microbit_poll.py might be the better example with `ubit.text ` replaced with `ubit.uart`.

Let me know if I've not explained that very well.

Svein-Tore, 2022-03-05:
Hi Barry!

I've been on your workshop (https://ukbaz.github.io/howto/ubit_workshop.html)
and there I found this program:

rom bluezero import microbit
ubit = microbit.Microbit(adapter_addr='xx:xx:xx:xx:xx:xx',
                         device_addr='yy:yy:yy:yy:yy:yy')
my_text = 'Hello, world'
ubit.connect()
while my_text is not '':
    ubit.text = my_text
    my_text = input('Enter message: ')

ubit.disconnect()

You wrote in your email that

So the
https://github.com/ukBaz/python-bluezero/blob/main/examples/microbit_poll.py
might
be the better example with ubit.text replaced with ubit.uart.
So I wonder, If  I change the line

 ubit.text=my_text

in the program above with
ubit.uart

Will it work  (be able to get my_text in bluetooth on data received)?

Best wishes, Svein-Tore

Svein-Tore, 2022-03-05:
Hi Barry!

After doing some testing I got it work!

Best wishes, Sve\in-Tore

ukBaz, 2022-03-06:
Well done on getting it working.  Sounds like this issue is resolved so I'll go ahead and close it. Please re-open if it is not resolved.

closed 2022-03-06 by ukBaz

Svein-Tore, 2022-03-08:
HI Barry!

Si I went into trouble.  I now get the following error:

***@***.***:~/.local/bin $ python3 testing_Micorbit.py
Traceback (most recent call last):
  File "/home/pi/.local/bin/testing_Micorbit.py", line 5, in <module>
    ubit.connect()
  File "/usr/local/lib/python3.9/dist-packages/bluezero/microbit.py", line
192, in connect
    self.ubit.connect()
  File "/usr/local/lib/python3.9/dist-packages/bluezero/central.py", line
89, in connect
    self.rmt_device.connect(timeout=timeout)
  File "/usr/local/lib/python3.9/dist-packages/bluezero/device.py", line
306, in connect
    raise dbus_exception
  File "/usr/local/lib/python3.9/dist-packages/bluezero/device.py", line
290, in connect
    self.bus.call_blocking(constants.BLUEZ_SERVICE_NAME,
  File "/usr/lib/python3/dist-packages/dbus/connection.py", line 652, in
call_blocking
    reply_message = self.send_message_with_reply_and_block(
dbus.exceptions.DBusException: org.bluez.Error.Failed: Software caused
connection abort

Any suggestion?

ukBaz, 2022-03-09:
There are not too many clues in that error message. It appears to be having an issue connecting. 
What did you change before it stopped working?
Did you change the code on the micro:bit? Do you need to delete the pairing information and re-pair the two devices?
Have you tried restarting the BlueZ daemon on the Linux machine (`sudo service bluetooth restart`)?

To get Bluetooth debug information on Linux:

- Are you able to connect to the device using `bluetoothctl`?
- Does `service bluetooth status` show errors?

When running your script have separate terminals open with the following running to get more debug information:

- `bluetoothctl`
- `journalctl -f -u bluetooth`
- `sudo busctl monitor org.bluez`
- `sudo btmon`

Svein-Tore, 2022-03-09:
Hi Barry!

Something went wrong with the pairing I guess!

Worked after pairing the microbit with the Rpi!

Svein-Tore