ukBaz/python-bluezero #374: scanning and listing devices without name throws dbus exception instead of returning None

userx14, 2022-07-14:
Hi,

When I try to print the name property from devices returned by iterating over device.Devices.available() I get the following exception for devices which do not have a name:
```dbus.exceptions.DBusException: org.freedesktop.DBus.Error.InvalidArgs: No such property 'Name'```

Instead of this error I would expect to get ```None``` in return.

https://github.com/ukBaz/python-bluezero/blob/1fb7a0e7df18b785fe073aa55cba1354633f6880/bluezero/device.py#L73-L77

https://github.com/ukBaz/python-bluezero/blob/1fb7a0e7df18b785fe073aa55cba1354633f6880/bluezero/dbus_tools.py#L386-L403

Is it possible that the wording of the error message has changed in dbus?
I'm using dbus v1.12.20-2 and bluez 5.55-3.1 on Kernel 5.10.0-15-amd64

Best,
Benjamin

ukBaz, 2022-07-14:
Hmm? It does seem like something has changed. I've done a google search but haven't been able to find any mention of it.

I've had a go at a fix on https://github.com/ukBaz/python-bluezero/tree/invalid_args branch.
Are you able to test it and let me know?

Testing for `InvalidArgs` seems like too broad an exception so I'm checking for `no such property` in the error message.

Thoughts and comments on the change welcome.

userx14, 2022-07-14:
@ukBaz ,

thank you for the fast fix, it solves the bug for me. 
Testing is a bit difficult, since it only seems to occur when a device from far away gets in range briefly,
but I managed to get some `None` device after 5 minutes and it works now without throwing an error.

Another problem I get when scanning after restarting the bluetooth service:
```
File "/usr/local/lib/python3.9/dist-packages/bluezero/device.py", line 216, in RSSI
    return self.remote_device_props.Get(
File "/usr/lib/python3/dist-packages/dbus/proxies.py", line 141, in __call__
    return self._connection.call_blocking(self._named_service,
File "/usr/lib/python3/dist-packages/dbus/connection.py", line 652, in call_blocking
    reply_message = self.send_message_with_reply_and_block(
dbus.exceptions.DBusException: org.freedesktop.DBus.Error.InvalidArgs: No such property 'RSSI'
```

Restarting my python tool doesn't help, but when I open the gnome bluetooth settings and open the "add device" menu the RSSI gets updated and the python tool no longer encounters this error.  Not sure if this is a bug caused by my specific hardware combination or if it is even worth fixing.

Thank you,
Benjamin

ukBaz, 2022-07-14:
The documentation at: https://git.kernel.org/pub/scm/bluetooth/bluez.git/tree/doc/device-api.txt has the following properties as optional:
```
		string Name [readonly, optional]
		string Icon [readonly, optional]
		uint32 Class [readonly, optional]
		uint16 Appearance [readonly, optional]
		array{string} UUIDs [readonly, optional]
		string Modalias [readonly, optional]
		int16 RSSI [readonly, optional]
		int16 TxPower [readonly, optional]
		dict ManufacturerData [readonly, optional]
		dict ServiceData [readonly, optional]
```
This probably means that any of these properties could not be on a device so need to use ` dbus_tools.get` rather than the dbus `Get` directly as they are doing.

e.g.
`return self.remote_device_props.Get(constants.DEVICE_INTERFACE, 'RSSI')`
should become
`return dbus_tools.get(self.remote_device_props, constants.DEVICE_INTERFACE, 'RSSI', None)`

I'll take a look at these later and update.

userx14, 2022-07-14:
My testing shows that both cases, dev.name == None and dev.RSSI == None have been fixed with your new commits. Thank you very much.

closed 2022-07-14 by userx14

ukBaz, 2022-07-14:
Thanks for testing.  There is a new release on PyPI with the fixes in.

https://pypi.org/project/bluezero/0.7.1/