ukBaz/python-bluezero #377: error 'br-connection-profile-unavailable' when trying heartbeat-monitor-central

aacebedo, 2022-07-28:
Hi

I am trying bluezero to build an app with two simple devices. One in a computer running arch, the other is a embedded board running debian sid.

Before doing a connection I was trying the Heartbeat-monitor example. One device acts as central the other as peripheral.
Discovery works but when the connection occurs, the central side returns the following error: 
```
Traceback (most recent call last):
  File "./heartrate_monitor_central.py", line 194, in <module>
    connect_and_run(hrm)
  File "./heartrate_monitor_central.py", line 153, in connect_and_run
    monitor.connect()
  File "/usr/lib/python3.10/site-packages/bluezero/central.py", line 91, in connect
    self.rmt_device.connect(timeout=timeout)
  File "/usr/lib/python3.10/site-packages/bluezero/device.py", line 303, in connect
    raise dbus_exception
  File "/usr/lib/python3.10/site-packages/bluezero/device.py", line 287, in connect
    self.bus.call_blocking(constants.BLUEZ_SERVICE_NAME,
  File "/usr/lib/python3.10/site-packages/dbus/connection.py", line 652, in call_blocking
    reply_message = self.send_message_with_reply_and_block(
dbus.exceptions.DBusException: org.bluez.Error.NotAvailable: br-connection-profile-unavailable
```

I can't find the solution for the moment. It does not depends on who the central or peripheral is.
However I can connect to the peripheral using an android app like LightBlue. So I guess something is wrong on the central code.

ukBaz, 2022-07-28:
When you say:

> It does not depends on who the central or peripheral is.

Do you mean that that you get the same error message on the same machine whether you are starting the peripheral or the central?

I noticed that the error message is suggesting you are running Python 3.10. I haven't tested the library with that version. 3.8 is the newest I've tested with.

To test the central, could you look at running the [BLE Peripheral Simulator](https://play.google.com/store/apps/details?id=io.github.webbluetoothcg.bletestperipheral) on an Android phone to see if your central device works with the Heart Rate on there.

If it doesn't work then letting me know some versions would be helpful.

`dbus-daemon --version` and `bluetoothd --version` are the two that would be most helpful

aacebedo, 2022-07-28:
> When you say:
> 
> 
> > It does not depends on who the central or peripheral is.
> 
> Do you mean that that you get the same error message on the same machine whether you are starting the peripheral or the central?
>

No I get the same error on the two machines. But only when they acts as central.

I'll try with the simulator but as I succeed in connecting to the peripheral with Android. 

Note that if I try to connect using dfeet or bluetoothctl I get the same error.

I'll give you the versions asap. On which version did you test?

ukBaz, 2022-07-28:
I have a Raspberry Pi with:
```
$ dbus-daemon --version
D-Bus Message Bus Daemon 1.12.20
$ python3 --version
Python 3.9.2
$ bluetoothd --version
5.55
```
And a Linux laptop with:
```
$ dbus-daemon --version
D-Bus Message Bus Daemon 1.12.16
$ python3 --version
Python 3.8.10
$ bluetoothd --version
5.53
```
If it is not working with `bluetoothctl` then it is at a lower level that Bluezero so I'm not sure I will be able to do too much

aacebedo, 2022-07-28:
I'll get the versions asap. Can you give your bluez config too?

ukBaz, 2022-07-28:
What do you mean by "bluez config"?

aacebedo, 2022-07-28:
Your `/etc/bluetooth/main.conf`

ukBaz, 2022-07-28:
Should be standard
```
$ cat /etc/bluetooth/main.conf 
[General]

# Default adaper name
# Defaults to 'BlueZ X.YZ'
#Name = BlueZ

# Default device class. Only the major and minor device class bits are
# considered. Defaults to '0x000000'.
#Class = 0x000100

# How long to stay in discoverable mode before going back to non-discoverable
# The value is in seconds. Default is 180, i.e. 3 minutes.
# 0 = disable timer, i.e. stay discoverable forever
#DiscoverableTimeout = 0

# How long to stay in pairable mode before going back to non-discoverable
# The value is in seconds. Default is 0.
# 0 = disable timer, i.e. stay pairable forever
#PairableTimeout = 0

# Automatic connection for bonded devices driven by platform/user events.
# If a platform plugin uses this mechanism, automatic connections will be
# enabled during the interval defined below. Initially, this feature
# intends to be used to establish connections to ATT channels. Default is 60.
#AutoConnectTimeout = 60

# Use vendor id source (assigner), vendor, product and version information for
# DID profile support. The values are separated by ":" and assigner, VID, PID
# and version.
# Possible vendor id source values: bluetooth, usb (defaults to usb)
#DeviceID = bluetooth:1234:5678:abcd

# Do reverse service discovery for previously unknown devices that connect to
# us. This option is really only needed for qualification since the BITE tester
# doesn't like us doing reverse SDP for some test cases (though there could in
# theory be other useful purposes for this too). Defaults to 'true'.
#ReverseServiceDiscovery = true

# Enable name resolving after inquiry. Set it to 'false' if you don't need
# remote devices name and want shorter discovery cycle. Defaults to 'true'.
#NameResolving = true

# Enable runtime persistency of debug link keys. Default is false which
# makes debug link keys valid only for the duration of the connection
# that they were created for.
#DebugKeys = false

# Restricts all controllers to the specified transport. Default value
# is "dual", i.e. both BR/EDR and LE enabled (when supported by the HW).
# Possible values: "dual", "bredr", "le"
#ControllerMode = dual

# Enables Multi Profile Specification support. This allows to specify if
# system supports only Multiple Profiles Single Device (MPSD) configuration
# or both Multiple Profiles Single Device (MPSD) and Multiple Profiles Multiple
# Devices (MPMD) configurations.
# Possible values: "off", "single", "multiple"
#MultiProfile = off

# Permanently enables the Fast Connectable setting for adapters that
# support it. When enabled other devices can connect faster to us,
# however the tradeoff is increased power consumptions. This feature
# will fully work only on kernel version 4.1 and newer. Defaults to
# 'false'.
#FastConnectable = false

[Policy]

# The ReconnectUUIDs defines the set of remote services that should try
# to be reconnected to in case of a link loss (link supervision
# timeout). The policy plugin should contain a sane set of values by
# default, but this list can be overridden here. By setting the list to
# empty the reconnection feature gets disabled.
#ReconnectUUIDs=00001112-0000-1000-8000-00805f9b34fb, 0000111f-0000-1000-8000-00805f9b34fb, 0000110a-0000-1000-8000-00805f9b34fb

# ReconnectAttempts define the number of attempts to reconnect after a link
# lost. Setting the value to 0 disables reconnecting feature.
#ReconnectAttempts=7

# ReconnectIntervals define the set of intervals in seconds to use in between
# attempts.
# If the number of attempts defined in ReconnectAttempts is bigger than the
# set of intervals the last interval is repeated until the last attempt.
#ReconnectIntervals=1, 2, 4, 8, 16, 32, 64

# AutoEnable defines option to enable all controllers when they are found.
# This includes adapters present on start as well as adapters that are plugged
# in later on. Defaults to 'false'.
AutoEnable=true
```

aacebedo, 2022-07-30:
Thank you.
It seems that if I force the le mode in this file it woks. However I cannot connect twice in a row. I get br-connection-abort-by-local the second tentative. I have to power off and power on the adpaters of the 2 machines to get it working again.

ukBaz, 2023-08-06:
closing because of inactivity

closed 2023-08-06 by ukBaz