ukBaz/python-bluezero #378: (question) Is it possible to pair and connect to a BT speaker with this library?

matyasf, 2022-08-03:
hi!

I'm trying to connect my Pi to a BT speaker; I've been writing code that parses the output of `bluetoothctl`, but its a mess.
Is it possible to do the following with this library:  
1. Scan for nearby BT devices.   
2. connect to one (no need for password/pin); specify it as a BT audio sink

I've looked trough the examples but could not find one that fits this use case, just the opposite (the code acts as a BT audio sink)
Thanks!

ukBaz, 2022-08-03:
Yes, writing code to interact with `bluetoothctl` is the wrong approach. Bluezero is is targeted at Bluetooth Low Energy so is not ideal for audio devices which are Bluetooth Classic

The adapter and device API at https://git.kernel.org/pub/scm/bluetooth/bluez.git/tree/doc will give you access to scanning and connecting to devices.
I am assuming that you are wanting to connect to UUID `0x110B` for Audio Sink. I would use something like https://pypi.org/project/pydbus/ library to interact with the D-Bus API.

Example of how to connect to a device is at:
https://stackoverflow.com/a/63751113/7721752
You don't need to worry about the GATT content, just the device connect.

matyasf, 2022-08-03:
thanks for the explanation!

closed 2022-08-03 by matyasf