ukBaz/python-bluezero #383: Feture request: Implement Agent for pairing support

Digitelektro, 2022-09-29:
Hi ukBaz!

I miss the pairing feature from this very nice library! If you don't mind adding this feature would be really helpful!
There is an example code which I tried and works great: [https://git.kernel.org/pub/scm/bluetooth/bluez.git/tree/test/simple-agent](https://git.kernel.org/pub/scm/bluetooth/bluez.git/tree/test/simple-agent)

In fact I already wrote a class based on this code, if you'd like I can share it with you, I think it just needs a little touch up and ready to use!

ukBaz, 2022-09-29:
Hi @Digitelektro,

I am very undecided on implementing the pairing agent within the library. It doesn't seem to offer a lot over and above what is available in `bluetoothctl`.

Pairing is typically a one-off setup activity so why not use `bluetoothctl`? If you think there is something I'm missing then I'm happy to take a look.

The balance with this library is keeping it manageable so that it can be tested when new releases are made. This library was intended to help people ramp up quickly on Python Bluetooth and that people would then move on if they needed something more specific.

Digitelektro, 2022-09-29:
Yes I agree on this is good for ramp up. I also use this project just for a POC, C++ implementation is on the way, but in the meantime I started to use this python lib as it is convinient and easy to make a working program very fast. I tought I can contribute to it and make it even better. This is the best python library for bluez out there, it is almost perfect! With this extra component it would be even better!

Sundareswar23, 2023-01-30:
> Hi ukBaz!
>
> I miss the pairing feature from this very nice library! If you don't mind adding this feature would be really helpful! There is an example code which I tried and works great: [...]

Please share the code, I need to cross verify

ukBaz, 2023-08-06:
I'm closing this because of inactivity

closed 2023-08-06 by ukBaz