ukBaz/python-bluezero #385: create 2 services from the same source?

David-Afik, 2022-10-12:
Hello, 
I'm trying to create 2 services in the same UUID 

so I have created a second service , and I want to use the same update_value function 

I don't want it to enter\run the function twice (one for each data - it's unnecessary)

is this a standard \ normal question ?

Thanks , 
 
***if we take the temperature example , 
I have created 2 random values in on the same read_value function - but what and how I need to return it ?

```
CPU_TMP_SRVC = '12341000-1234-1234-1234-123456789abc'
CPU_TMP_CHRC = '12341002-1234-1234-1234-123456789abc'
CPU_TMP2_CHRC = '12341003-1234-1234-1234-123456789abc'

def read_value():

    cpu_value = random.randrange(3200, 5310, 10) / 100
    cpu_temp = random.randrange(1, 10000, 48)

    return list(int(cpu_value * 100).to_bytes(2,
                                              byteorder='little', signed=True))

def update_value(characteristic):
    # read/calculate new value.
    new_value = read_value()
 

    # Causes characteristic to be updated and send notification
    characteristic.set_value(new_value)
    print ('new value is ' , new_value)
    sys.stdout.flush()
     return characteristic.is_notifying

def notify_callback(notifying, characteristic):
    if notifying:
        async_tools.add_timer_seconds(2, update_value, characteristic)

def main (adapter_address):
 cpu_monitor.add_service(srv_id=1, uuid=CPU_TMP_SRVC, primary=True)
    # Add characteristic
    cpu_monitor.add_characteristic(srv_id=1, chr_id=1, uuid=CPU_TMP_CHRC,
                                   value=[], notifying=False,
                                   flags=['read', 'notify'],
                                   read_callback=read_value,
                                   write_callback=None,
                                   notify_callback=notify_callback
                                   )
cpu_monitor.add_characteristic(srv_id=1, chr_id=2, uuid=CPU_TEMP2_CHRC,
                                   value=[], notifying=False,
                                   flags=['read', 'notify'],
                                   read_callback=read_value,
                                   write_callback=None,
                                   notify_callback=notify_callback
                                   )

```

Thanks ,

ukBaz, 2022-10-12:
If I have understood correctly then your `read_value` function needs to look something like this:

```python
def read_value():
    cpu_value = random.randrange(3200, 5310, 10)
    cpu_temp = random.randrange(1, 10000, 48)
    return struct.pack('<hh', cpu_value, cpu_temp)
```

Take a look at the [GATT Specification Supplement](https://www.bluetooth.com/specifications/assigned-numbers/). Lots of good stuff in there about characteristics. The sections I would suggest to start with are:

> 2.3 Values and represented values .................................................................................................. 16
> 2.3.1 Interpretation of values .................................................................................................................... 16
> 2.3.2 Scalar values ................................................................................................................................... 16
> 2.4 Octet and bit ordering ................................................................................................................. 17
> 2.5 Endianness ................................................................................................................................. 17

```

David-Afik, 2022-10-30:
I will try 

Thank you

ukBaz, 2023-08-06:
Closing because of inactivity

closed 2023-08-06 by ukBaz