ukBaz/python-bluezero #387: unable to send data on python 3.7.3 (but on 3.10 all works)?

David-Afik, 2022-10-30:
Hello, 
I'm trying to send simple "0" \ "1" to my remote device 
when I run the code on windows with Python 3.10  - it's working 
when I run the same code on Linux with Python 3.7.3 - it's now working 
I don't get and error - but I can see nothing is send 
what am I missing ? 
```
 New_Led_Status = input('''
     Enter new led status:
1 - Led on ,  will turn on the  relay 
0 - led off , will turn off the relay 

        '''
 try:
                    value = await client.read_gatt_char(Led_Char)  # read old value
                    print("Pre-Write Value: {0}".format(value))
                    if New_Led_Status == "1":
                        await client.write_gatt_char(Led_Char, b'\01', response=False)  # send new value
                    elif New_Led_Status == "0":
                        await client.write_gatt_char(Led_Char, b'\00', response=False)  # send new value
                    else:
                        await client.write_gatt_char(Led_Char, str.encode(New_Led_Status), response=False)  # anything else
                    value = await client.read_gatt_char(Led_Char)
                    print("Post-Write Value: {0}".format(value))
```

on windows  I get this :
```
     Enter new led status:
1 - Led on ,  will turn on the  relay 
0 - led off , will turn off the relay 
1

Pre-Write Value: bytearray(b'8')
Post-Write Value: bytearray(b'\x01')
finish
```
on Linux I get this : 
```
     Enter new led status:
1 - Led on ,  will turn on the  relay
0 - led off , will turn off the relay
0

Pre-Write Value: bytearray(b'\x01')
Post-Write Value: bytearray(b'\x01')
finish
```

ukBaz, 2022-10-30:
Hi @David-Afik ,

Did you open this issue on the wrong repository?  Bluezero doesn't run on Windows and doesn't need `await` commands because it doesn't use asyncio.

Did you mean to open this on the `bleak` repo?

Having said that, do you need `b"\x00"` rather than `b"\00"`?

David-Afik, 2022-10-30:
really ?  very strange 
because I'm running the code with 
```
import asyncio
from bleak import BleakScanner, BleakClient
```
on my windows 10 PyCharm - and it's working
I'm also reading values - and everything is OK 

I have try use 
```
if New_Led_Status == "0":
    await client.write_gatt_char(Led_Char, b'\x00'
if New_Led_Status == "1":
    await client.write_gatt_char(Led_Char, b'\x01')
```)  
got the same result , didn't send anything

ukBaz, 2022-10-30:
Thanks for confirming it is `bleak` you are using. I'll go ahead and close this as nothing the `bluezero` repo can do to help.

I guess it wasn't an issue `\x`. I'll let you work it out with the bleak team.

closed 2022-10-30 by ukBaz