ukBaz/python-bluezero #390: Disabling BlueZ request to pair with central on connect.

WayneKeenan, 2022-11-30:
Hello,   

tl;dr:   Found a way to stop the pairing request, but what's the most correct and simplest way to disable the BlueZ pairing request from bluezero/Python code?

```
Environment:
BLE Client: LightBlue app on iOS 16.1.1, 
Pi:         Pi4  - Stretch - 2021-05-07
BlueZ:      5.50
Python:     3.7.3
Code:       heartrate_peripheral.py
```

Well, this is new (to me, today) behaviour. 
I guess BlueZ is trying to be 'more secure' by default, sensible, but kinda annoying...

```
> ACL Data RX: Handle 65 flags 0x02 dlen 11                                                                               #118 [hci0] 16.967848
      SMP: Pairing Request (0x01) len 6
        IO capability: KeyboardDisplay (0x04)
        OOB data: Authentication data not present (0x00)
        Authentication requirement: Bonding, No MITM, SC, No Keypresses, CT2 (0x29)
        Max encryption key size: 16
        Initiator key distribution: EncKey IdKey LinkKey (0x0b)
        Responder key distribution: EncKey IdKey LinkKey (0x0b)
```

The issue I have elsewhere is that the recommended React Native JS client library doesn't support pairing, joy. 

So, until that library gets replaced with something sensible I found a way to disable the request:

```
sudo btmgmt
pairable off
```

Although it appears to work on the surface there are 2 issues:

1. The setting doesn't survive a reboot (need to fathom/hope which Bluez config setting it is)
2. There are lots of repeated entries of the form:
```
> ACL Data RX: Handle 65 flags 0x02 dlen 11                                                                               #245 [hci0] 31.215209
      SMP: Pairing Request (0x01) len 6
        IO capability: KeyboardDisplay (0x04)
        OOB data: Authentication data not present (0x00)
        Authentication requirement: Bonding, No MITM, SC, No Keypresses, CT2 (0x29)
        Max encryption key size: 16
        Initiator key distribution: EncKey IdKey LinkKey (0x0b)
        Responder key distribution: EncKey IdKey LinkKey (0x0b)
@ MGMT Event: Authentication Failed (0x0011) plen 8                                                                   {0x0003} [hci0] 31.215289
        LE Address: 72:0A:05:DA:BC:A5 (Resolvable)
        Status: Authentication Failed (0x05)
```

Regards
Wayne

ukBaz, 2022-12-04:
Can I check my understanding of what the problem is here... Are you saying that you've created a peripheral with Bluezero and when you connect to it from a central device it asks for pairing even though the Bluezero created peripheral has no requirement for pairing?

Is this the "battery" plugin issue?
https://github.com/ukBaz/python-bluezero/issues/336#issuecomment-912915781

WayneKeenan, 2022-12-04:
Correct and setting `ReverseServiceDiscovery` to `true` didn't work for me.

ukBaz, 2022-12-04:
Would removing the Battery plugin be a better solution for you?
```
/usr/lib/bluetooth/bluetoothd -P battery
```

WayneKeenan, 2022-12-04:
perfect, thanks, that worked.

WayneKeenan, 2022-12-04:
also just to note that there's no more pairing request 'noise' in the logs.

closed 2022-12-04 by WayneKeenan