ukBaz/python-bluezero #391: AttributeError: module 'bluezero.async_tools' has no attribute 'EventLoop'

onurcoskun14, 2023-01-23:
When I try to run cpu_temperature.py I get this error message

Traceback (most recent call last):
  File "/home/pi/deneme.py", line 68, in <module>
    main(list(adapter.Adapter.available())[0].address)
  File "/home/pi/.local/lib/python3.9/site-packages/bluezero/adapter.py", line 55, in available
    yield Adapter(adapter['Address'])
  File "/home/pi/.local/lib/python3.9/site-packages/bluezero/adapter.py", line 86, in __init__
    self.mainloop = async_tools.EventLoop()
AttributeError: module 'bluezero.async_tools' has no attribute 'EventLoop'

ukBaz, 2023-01-23:
I am not able to reproduce this. I've done a test as the following:
```none
$ cat /etc/*-release
PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
NAME="Raspbian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
$ systemctl status bluetooth
● bluetooth.service - Bluetooth service
     Loaded: loaded (/lib/systemd/system/bluetooth.service; enabled; vendor preset: enabled)
     Active: active (running) since Mon 2022-12-19 19:26:43 GMT; 1 months 4 days ago
       Docs: man:bluetoothd(8)
   Main PID: 625 (bluetoothd)
     Status: "Running"
      Tasks: 1 (limit: 1598)
        CPU: 1.461s
     CGroup: /system.slice/bluetooth.service
             └─625 /usr/libexec/bluetooth/bluetoothd
$ pip install bluezero
Defaulting to user installation because normal site-packages is not writeable
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting bluezero
  Using cached https://www.piwheels.org/simple/bluezero/bluezero-0.7.1-py2.py3-none-any.whl (41 kB)
Installing collected packages: bluezero
Successfully installed bluezero-0.7.1
$ python cpu_temperature.py 
CPU temperature is 42.6°C
Advertisement registered
```

Are you using RPi OS lite or anything like that?
__

onurcoskun14, 2023-01-23:
I'am using Rpi zero w

ukBaz, 2023-01-23:
Being on a RPi Zero W shouldn't be a problem.   

Did `systemctl status bluetooth` work for you?
Does running `bluetoothctl show` on the command line work?

Should give you somwthing like:
```none
$ bluetoothctl show
Controller B8:27:EB:22:57:E0 (public)
        Name: SensePi
        Alias: SensePi
        Class: 0x002c0000
        Powered: yes
        Discoverable: no
        DiscoverableTimeout: 0x000000b4
        Pairable: no
        UUID: A/V Remote Control        (0000110e-0000-1000-8000-00805f9b34fb)
        UUID: Audio Source              (0000110a-0000-1000-8000-00805f9b34fb)
        UUID: PnP Information           (00001200-0000-1000-8000-00805f9b34fb)
        UUID: Audio Sink                (0000110b-0000-1000-8000-00805f9b34fb)
        UUID: Headset                   (00001108-0000-1000-8000-00805f9b34fb)
        UUID: A/V Remote Control Target (0000110c-0000-1000-8000-00805f9b34fb)
        UUID: Generic Access Profile    (00001800-0000-1000-8000-00805f9b34fb)
        UUID: Generic Attribute Profile (00001801-0000-1000-8000-00805f9b34fb)
        UUID: Device Information        (0000180a-0000-1000-8000-00805f9b34fb)
        UUID: Headset AG                (00001112-0000-1000-8000-00805f9b34fb)
        Modalias: usb:v1D6Bp0246d0537
        Discovering: no
        Roles: central
        Roles: peripheral
Advertising Features:
        ActiveInstances: 0x00 (0)
        SupportedInstances: 0x05 (5)
        SupportedIncludes: tx-power
        SupportedIncludes: appearance
        SupportedIncludes: local-name
```

onurcoskun14, 2023-01-24:
[image]

It seems okey

onurcoskun14, 2023-01-24:
[image]

[image]

I couldn't solve the Failed to set privacy error

onurcoskun14, 2023-01-27:
By the way I'am using OS Lite is that a problem ?

ukBaz, 2023-01-27:
> By the way I'am using OS Lite is that a problem ?

Not a problem, no.   I haven't stayed up to date with the changes on OS Lite, but there has historically been some extra steps that are needed. e.g. https://raspberrypi.stackexchange.com/a/116843/121848

The most worrying of what you've shown is the `--compat` option.  

The issues you have are system specific and seem to be a result of you customising your system. This means it is not Bluezero specific and are difficult to diagnose without access to your system. I think this means it is going to be difficult for me to help you.

onurcoskun14, 2023-01-27:
I have installed RPİ OS Full and it works now. OS Lite might be the problem.
[image]
this one from OS Lite

----------------------------

[image]
This one from OS Full

ukBaz, 2023-01-27:
Thanks for letting me know the update. Good news.

I'll go ahead and close this.

closed 2023-01-27 by ukBaz