ukBaz/python-bluezero #394: Can't install dbus-python to use bluezero

bernardolansing, 2023-03-10:
I am trying to use this library inside a Raspberry Zero W. I've sucessfully installed it using `pip install bluezero`, but if I import it, it will cause the following error:
`
Traceback (most recent call last):
  File "/home/bernardo/projects/anymais_firmware/src/bluetooth.py", line 1, in <module>
    from bluezero import adapter
  File "/home/bernardo/projects/anymais_firmware/venv/lib/python3.11/site-packages/bluezero/adapter.py", line 4, in <module>
    import dbus
ModuleNotFoundError: No module named 'dbus'
`

Well, so I try to install dbus-python, and now I got a kilometer long error report, listing a lot of missing C libraries:

`
 [32/32] Linking target test/test-import-repeatedly
      FAILED: test/test-import-repeatedly
      cc  -o test/test-import-repeatedly test/test-import-repeatedly.p/import-repeatedly.c.o -Wl,--as-needed -Wl,--no-undefined -Wl,--start-group /usr/local/lib/libpython3.11.a -Wl,--end-group
      /usr/bin/ld: /usr/local/lib/libpython3.11.a(faulthandler.o): in function `faulthandler_thread':
      /home/bernardo/Downloads/Python-3.11.0/./Modules/faulthandler.c:609: undefined reference to `pthread_sigmask'
      /usr/bin/ld: /usr/local/lib/libpython3.11.a(floatobject.o): in function `_float_div_mod':
      /home/bernardo/Downloads/Python-3.11.0/Objects/floatobject.c:675: undefined reference to `fmod'
      /usr/bin/ld: /usr/local/lib/libpython3.11.a(floatobject.o): in function `float_rem':
      /home/bernardo/Downloads/Python-3.11.0/Objects/floatobject.c:655: undefined reference to `fmod'
      /usr/bin/ld: /usr/local/lib/libpython3.11.a(floatobject.o): in function `float___round___impl':
      /home/bernardo/Downloads/Python-3.11.0/Objects/floatobject.c:1090: undefined reference to `round'
      /usr/bin/ld: /home/bernardo/Downloads/Python-3.11.0/Objects/floatobject.c:1093: undefined reference to `round'

`
(list goes on)

If that is important, I am using Python 3.11 inside a virtualenv. Perhaps this issue is not related to bluezero itself, but only with dbus-python, but that repository is not open to issues and I could not find anyone facing such trouble. I would appreciate anyone willing to lend me a hand.

ukBaz, 2023-03-10:
I haven't tested with Python 3.11, but normally the issue is with Virtual Environments and that D-Bus and the library for the event loop.

The addition steps are documented over in this other ticket:

https://github.com/ukBaz/python-bluezero/issues/307#issuecomment-1426706775

wallem89, 2023-03-30:
@bernardolansing have you tried installing it with:

```
sudo apt-get install python3-dbus
```

I had a similar issue and found this https://stackoverflow.com/a/37927053. Mind the comment on this answer that it should be ```python3-dbus``` for Python 3. Then it worked for me!