ukBaz/python-bluezero #402: Advertisment of peripheral

furgerf, 2024-01-18:
Hi, I'm trying to create a GATT server on a peripheral device - disclaimer: I'm new to BLE and may well misunderstand something 🙂

Empirically, I find that I can only connect find the device in a LEscan and connect to it if it's advertising (but discoverable doesn't seem to matter(?)).
I see that instantiating a `Peripheral` automatically makes it discoverable (before eventually publishing the advertisements) - but the device isn't made advertising. I couldn't find methods to activate that explicitly.

So I'm wondering what the intended behavior is - should advertising be activated automatically as well (like discoverability)? Should advertising not be necessary? Should I handle this some other way?

Some additional info - when I boot the device, I see the following with `btmgmt`:
```
[mgmt]# info
Index list with 1 item
hci0:   Primary controller
        addr XX:XX:XX:XX:XX:XX version 9 manufacturer 305 class 0x000000
        supported settings: powered connectable fast-connectable discoverable bondable link-security ssp br/edr le advertising secure-conn debug-keys privacy static-addr
        current settings: powered ssp br/edr le secure-conn
        name hostname
        short name
```

After starting the application, which creates the `Peripheral` and adds some services/characteristics and then publishes:
```
[mgmt]# info
Index list with 1 item
hci0:   Primary controller
        addr XX:XX:XX:XX:XX:XX version 9 manufacturer 305 class 0x100000
        supported settings: powered connectable fast-connectable discoverable bondable link-security ssp br/edr le advertising secure-conn debug-keys privacy static-addr
        current settings: powered connectable discoverable ssp br/edr le secure-conn
        name hostname
        short name
```

Note how it's now connectable and discoverable but not advertising.

Using bluezero 0.8.0 with bluez 5.53.

Any assistance is greatly appreciated 👍

ukBaz, 2024-01-18:
The target of the Bluezero library is to put a facade to help get started with the BlueZ D-Bus API.  To look behind that facade at what is actually going on; For the creation of a BLE `Peripheral`, an advertisement and a GATT server needs to be created and published to the Bluetooth daemon (`bluetoothd`) running on the Linux machine. The `Peripheral` is asynchronous so it will only appear when the event loop (`mainloop`) is run. When that is run then `bluetoothd` will make them available over the air. `bluetoothd` will not advertise when there is a connection to the GATT server.
Advertising will not work without the adapter being discoverable. 

There isn't a great deal of information in your text for me to speculate what might be going wrong as there is no code and little debug information. Have your tried running one of the examples in the repo?
If I were to speculate based on what I've seen before, then either the event loop is not being started or your advertisement is too big.

When running your script have separate terminals open with the following running to get more debug information:

-     bluetoothctl
-     journalctl -f -u bluetooth
-     sudo busctl monitor org.bluez
-     sudo btmon

If you do a `show` inside `bluetoothctl` then it will report how many advertisements are registered and how many are active.

If it looks to be running without error, then use a generic Bluetooth Low Energy scanning and exploration tool, such as [nRF Connect](https://www.nordicsemi.com/Products/Development-tools/nrf-connect-for-mobile), to see what is being advertised and connect to the GATT server.

ukBaz, 2024-01-19:
Glad to hear you are making progress.

> There's another thing, I can read `adapter.discoverabletimeout` but when I try to write it, I get a `DBusException("Invalid signature for 'DiscoverableTimeout'")`. Setting `discoverable-timeout` manually with bluetoothctl works.
Is there specific information I can provide and should I open another issue for this?

It should be as simple as:
```python
adapter.discoverabletimeout = 180
```
If that isn't working then open a new issue and we can work on it there.

closed 2024-01-21 by furgerf