ukBaz/python-bluezero #408: Setting Manufacturer data in Advertiser only works if data is less than 3 bytes long

maticsr, 2024-04-22:
Hello,

I have a case where I need to set the manufacturer data to a 16 bytes constant. 

The issue with bluezero is that setting manufacturer data using this function:

`bridge.advert.manufacturer_data(77, [0x00, 0x01, 0x02, 0x03])` only works if the data array is 2 bytes or less. If the array is longer, the following error message is received: Failed to register advertisement: org.bluez.Error.Failed: Failed to parse advertisement.

Is this a bug or am I doing something wrong?

ukBaz, 2024-04-23:
BLE advertisements have very limited space so my speculation is that you are running out of space. You can get more debug information by running the following in separate terminals:

-     bluetoothctl
-     journalctl -f -u bluetooth
-     sudo busctl monitor org.bluez
-     sudo btmon

maticsr, 2024-04-23:
Thank you for a very prompt response. Running out of space was my first thought as well, I just didn't know how to approach it. Thanks for the debugging hint. Will test tonight.

ukBaz, 2024-04-23:
You may also find https://github.com/ukBaz/python-bluezero/issues/355 a useful reference.

closed 2024-04-23 by maticsr