ukBaz/python-bluezero #409: How to show Txpower in advertisement data

21317389, 2024-05-05:
HELLO,
I can't set the TXPOWER in advertisement.
How to solve it, and I think maybe https://github.com/ukBaz/python-bluezero/issues/333 can solve me problem.
But I can't understand how to fix.

Sincerely.

21317389, 2024-05-06:
Thank you for your prompt reply.
I want to generate the txpower in my advertisement data , just like this red circle at the photo.
[image]

And my code is modified from the [ble_uart.py](https://github.com/ukBaz/python-bluezero/blob/main/examples/ble_uart.py).
I just create the local name and service UUID.

And, I didn't get the error messages

ukBaz, 2024-05-06:
> And my code is modified from the [ble_uart.py](https://github.com/ukBaz/python-bluezero/blob/main/examples/ble_uart.py).

And how have you modified it to include the TX power?

21317389, 2024-05-06:
Oh, not yet.
It is just the example.
This is my goal, but I have no idea how to finish this part.

ukBaz, 2024-05-07:
Can you check that `tx-power` is supported on your setup. If you do the following:
```shell
busctl get-property org.bluez /org/bluez/hci0 org.bluez.LEAdvertisingManager1 SupportedIncludes
```

Do you get:
```text
as 3 "tx-power" "appearance" "local-name"
```

21317389, 2024-05-08:
YES!
I actually get those item!
But those item just means I can transfer them, right?

And, I also reference this https://github.com/ukBaz/python-bluezero/issues/323. I thought this proble is similar with mine.

However, I add ble_uart.advert.IncludeTxPower = 'True' in my code, the operation did not result in any action.

ukBaz, 2024-05-09:
> > Do you get:
> > `as 3 "tx-power" "appearance" "local-name"`

> YES!
> I actually get those item!
> But those item just means I can advert them, right?

You are correct, it only means that those values can be added to the advert.

What #333 is discussing is that the BlueZ API was changed and that Bluezero hasn't been updated to reflect that. So the change you made to your code won't work unless the library is updated.
I started to look at doing the update when you opened this issue but ran out of time and it might be a while before I can get back to this. It's not helped by the fact that my normal development machine doesn't support `tx-power` so I need to move to a different machine.

ukBaz, 2024-05-10:
To include some information here on how the BlueZ API has changed. In the 5.45 release `TX power` was included with the boolean property `IncludeTxPower` on interface `org.bluez.LEAdvertisement1`

This is documented at: https://git.kernel.org/pub/scm/bluetooth/bluez.git/tree/doc/advertising-api.txt?h=5.45#n64

```text
		bool IncludeTxPower

			Includes the Tx Power in the advertising packet.
			If missing, the Tx Power is not included.
```

In more recent versions of the API how `TX power` is included has changed. They removed the previous `IncludeTxPower` property. The API now requires the string `tx-power` in the array property of `Includes`

This is documented at: https://git.kernel.org/pub/scm/bluetooth/bluez.git/tree/doc/org.bluez.LEAdvertisement.rst?h=5.73#n123

```text
array{string} Includes

	List of features to be included in the advertising packet.

	Possible values:

	See **org.bluez.LEAdvertisingManager(5)** **SupportedIncludes**
	property.
```

For reference, `SupportedIncludes` is documented at:
https://git.kernel.org/pub/scm/bluetooth/bluez.git/tree/doc/org.bluez.LEAdvertisingManager.rst?h=5.73#n80

```text
array{string} SupportedIncludes [readonly]

	List of supported system includes.

	Possible values:

	:"tx-power":
	:"appearance":
	:"local-name":
	:"rsi":
```

This is the change that needs to be reflected in Bluezero.

ukBaz, 2024-05-12:
I've created a feature branch where I've made the code changes:\
https://github.com/ukBaz/python-bluezero/tree/tx_pwr_ad

The code I used to test this was:
```python
from bluezero import advertisement
from bluezero import constants
advert_id = 1
ad_type = 'broadcast'
ad_mngr_options = {}
beacon = advertisement.Advertisement(advert_id, ad_type)
beacon.include_tx_power = True
print(beacon.GetAll(constants.LE_ADVERTISEMENT_IFACE))
beacon.local_name = "MyBeacon"
beacon.service_UUIDs = ['FEAA']
beacon.service_data = {'FEAA': [0x10, 0x08, 0x03, 0x75, 0x6B, 0x42, 0x61, 0x7A, 0x2e, 0x67, 0x69, 0x74, 0x68, 0x75, 0x62, 0x2E, 0x69, 0x6F]}
ad_manager = advertisement.AdvertisingManager()
ad_manager.register_advertisement(beacon, ad_mngr_options)
beacon.start()
```
In nRF Connect app on the phone I scanned for the beacon. 

With the `include_tx_power` being set to `False` I got:
[image]

With the `include_tx_power` being set to `True` I got:
[image]

This looks to be what you are expecting @21317389 ?
Can you test the `tx_pwr_ad` branch and let me know if it works for you. If it appears to be working I'll merge it in to main and do a release.

21317389, 2024-05-12:
II'm really grateful for your help, and I think I'll run some tests after I finish this work segment. Thank you so much for your assistance.
Sincerely!!!!

closed 2024-05-14 by 21317389

ukBaz, 2024-05-18:
Added the 0.9.0 release to PyPI that contains this fix
https://pypi.org/project/bluezero/

PR #410 (2024-05-18): Add capability to include tx power in advertisement
Fix #409