ukBaz/python-bluezero #414: Version 0.9.0 cannot be installed via pip

erichiggins, 2024-09-09:
Hello,

The build system for my project, which depends on bluezero, started failing back in May when version 0.9.0 was released. The issue seems related to the new dependencies that were added, which cannot be installed on Linux systems for some reason.

To resolve the issue, I've had to pin to version 0.8.0.

The following is the error log from the build, which runs on Github actions, and I was able to reproduce locally in a clean Python 3.12 virtual environment by running `pip install bluezero`.

```
Collecting PyGObject (from bluezero)
  Using cached pygobject-3.48.2.tar.gz (715 kB)
  Installing build dependencies: started
  Installing build dependencies: finished with status 'error'
  error: subprocess-exited-with-error
  
  × pip subprocess to install build dependencies did not run successfully.
  │ exit code: 1
  ╰─> [52 lines of output]
      Collecting meson-python>=0.12.1
        Using cached meson_python-0.16.0-py3-none-any.whl.metadata (4.1 kB)

...
 -- abbreviated --
...
            Package cairo was not found in the pkg-config search path.
            Perhaps you should add the directory containing `cairo.pc'
            to the PKG_CONFIG_PATH environment variable
            No package 'cairo' found
            Command '['pkg-config', '--print-errors', '--exists', 'cairo >= 1.15.10']' returned non-zero exit status 1.
            [end of output]
      
        note: This error originates from a subprocess, and is likely not a problem with pip.
        ERROR: Failed building wheel for pycairo
      Failed to build pycairo
      ERROR: ERROR: Failed to build installable wheels for some pyproject.toml based projects (pycairo)
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: subprocess-exited-with-error

× pip subprocess to install build dependencies did not run successfully.
│ exit code: 1
╰─> See above for output.

note: This error originates from a subprocess, and is likely not a problem with pip.
Error: Process completed with exit code 1.
```

ukBaz, 2024-09-09:
What operating system are you using this on?
And have you done any research as to what a solution might be for this? (as there is a requirement for PyGObject to be installed)

erichiggins, 2024-09-09:
Both Linux systems are running Ubuntu (version 20 or 22 I believe).
I have not investigated solutions (most of my time was spent fixing the broken build) -- it's not clear to me why PyGObject is required or how it's used.

ukBaz, 2024-09-09:
PyGObject provides GLib and the async mainloop

erichiggins, 2024-09-09:
I can try to help but I need to know more. Why is GLib needed?

ukBaz, 2024-09-09:
I'm not quite sure I understand what you are looking for. I'll give some answers but please let me know if I have misunderstood what you want.

You might want to start with the instructions for how to install the package. https://pygobject.gnome.org/getting_started.html#ubuntu-logo-ubuntu-debian-logo-debian

As to why it is needed, a lot of Bluetooth functionality is asynchronous (e.g. Characteristics notifications or scanning for devices) and the BlueZ library (the official Bluetooth stack on Linux) that Bluezero builds on top of uses the GLib library.  The async functionality comes from the "mainloop". More information at: https://docs.gtk.org/glib/main-loop.html

erichiggins, 2024-09-09:
This helps some, thank you!

Perhaps my question should be, what does this library provide that core Python and the standard libraries do not? Async functionality is available within Python 3, for example.
https://docs.python.org/3/library/asyncio.html

Looking at the diff between version 0.8.0 and 0.9.0, it doesn't appear that much has functionally changed.
https://github.com/ukBaz/python-bluezero/commit/8e984b62508b2feb6b953790393b3c0630a8eb64

Was adding this dependency necessary in order to support the addition of tx-power in advertising requests?

WayneKeenan, 2024-09-09:
PyObject is a Python C Extension module that provides access to the API's for the native C/C++ world of: GObjects, GLib mainloop, events etc.

Using Python's asyncio would still require a way to call the native C API's, i.e. PyObject.

The architecture: https://pygobject.gnome.org

WayneKeenan, 2024-09-09:
Perhaps this might help:

```
apt install libcairo2-dev
```

or this (which depends on the above)
```
apt install libcairo-gobject2             
```

FYI:

Using Python's asyncio would still require a way to call the native C API's for the GLib runtime that the PyGObject C Extension Module provides access to.

The architecture: https://pygobject.gnome.org/

WayneKeenan, 2024-09-09:
I think adding PyGObject brings in 1 or more dependancies that would sometimes not be available in a venv, but it's been a while.

ukBaz, 2024-09-10:
Hi @WayneKeenan ,

> I think explicitly adding PyGObject brings in 1 or more dependancies that would sometimes not be available in a venv, but it's been a while so please don't quote me.

I believe that the PyGObject project has done some work to make things more pip friendly these days and that is why I added the dependency. However, testing on different systems and different OSes is difficult for me. I'm hoping the more eyes we can get on this the more robust this will become over time 🤞

WayneKeenan, 2024-09-10:
> Both Linux systems are running Ubuntu (version 20 or 22 I believe). I have not investigated solutions (most of my time was spent fixing the broken build) -- it's not clear to me why PyGObject is required or how it's used.

What CPU arch are you on?

erichiggins, 2024-09-10:
> Hi @erichiggins ,
> 
> > Both Linux systems are running Ubuntu (version 20 or 22 I believe).
> 
> What CPU arch are you on?
> 
> Is it Ubuntu Server or Desktop edition?

For mine, Intel arch, Ubuntu Desktop (variant).
Unsure what the arch/version is for the Github action runner but I can find out.

WayneKeenan, 2024-09-11:
I managed to replicate the error on a fresh AWS EC2 Ubuntu 24.04 LTS instance; a long story involving not have much success running VM's on my M1 Mac and also no RPi to hand.  

But, the steps below gets to the point of being able to successfully install `bluezero` and run the examples, although it ultimately timeouts with an `DBus.Error.NoReply` as `bluez` doesn't respond as there is no underlying BLE hardware.

The critical thing is to remove PyGObject from `setup.py`.

```
sudo apt install python3.12-venv
sudo apt  install bluez        
python3 -m venv --system-site-packages venv
. ./venv/bin/activate
wget https://github.com/ukBaz/python-bluezero/archive/refs/tags/v0.9.0.tar.gz
tar xf v0.9.0.tar.gz 
cd python-bluezero-0.9.0/
nano setup.py  #   edit line 5 to be:  required_packages = []
pip install .  
python3 examples/eddystone_scanner.py
```

Hope it helps from both a user and maintainer perspective.

WayneKeenan, 2024-09-11:
To resolve the original error I did this:

`sudo  apt install libcairo-dev`

However, there is a further PyGObject dependancy on `python` (version 2) which I could not resolve.

Jordan-Jarvis, 2024-12-13:
@WayneKeenan I was able to resolve the issue using 
`sudo apt install libgirepository1.0-dev`

> To resolve the original error I did this:
> 
> `sudo apt install libcairo-dev`
> 
> However, there is a further PyGObject dependancy on `python` (version 2) which I could not resolve.

sunil390, 2025-03-16:
> I'm not quite sure I understand what you are looking for. I'll give some answers but please let me know if I have misunderstood what you want.
>
> You might want to start with the instructions for how [...]

Latest  pygobject-3.52.2 has a dependency on GIRepository-2.0 which is not available for latest raspberry pi os
sudo apt install libgirepository-2.0-dev  is failing 
```
zpi@pi:~ $ sudo apt install libgirepository-2.0-dev
[sudo] password for zpi:
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package libgirepository-2.0-dev
E: Couldn't find any package by glob 'libgirepository-2.0-dev'
```
Untill this is made available I am not able to get bluezero installed by the currently documented process.

ukBaz, 2025-03-23:
> Until this is made available I am not able to get bluezero installed by the currently documented process.

Which "documented process" are you referring to?

I have had a look at this and it seems to be the version of PyGObject (3.52.3) on PyPI that is the issue
https://pypi.org/project/PyGObject/#history

I have done my experimentation on Debian 12 (which RPi OS is built on top of) and it has the same issue.

Looking at the release notes they moved to `girepository-2.0` in version `3.52.0`
https://gitlab.gnome.org/GNOME/pygobject/-/blob/main/NEWS

If I install a version less than that then bluezero seems to install successfully for me.

This was my test:
```
python3 -m venv venv
. ./venv/bin/activate
pip install --upgrade "PyGObject<3.52.0"
pip install bluezero
```
Which probably means a patch release is required to change the following to not install the latest version

https://github.com/ukBaz/python-bluezero/blob/90f38a0fefcd3d4f4e00fbb4af76bd132a0e74f4/setup.py#L16

ukBaz, 2025-03-23:
Version 0.9.1 released:
https://pypi.org/project/bluezero/0.9.1/