ukBaz/python-bluezero #416: How to change manufacterer data after an advertisement has been registered?

AKFTHEGREAT, 2024-10-04:
Hi, I have recently had some issues attempting to change a device's manufacturer data after the advertisement has been registered.
Currently, I have only been able to establish manufacturer data on an advert before it is published, which works just fine.
However, it seems that it isn't possible to change manufacturer data after the peripheral has been published. 

Naturally, this leads me to believe that one must stop advertising, update the manufacturer data and then restart advertising, but I am unable to even stop advertising at all. (

Currently running the code on an rpi0 using tinycore linux. I hope this isn't a hardware/software issue with my setup as even trying to run multiple services on this device seems not to work, despite no hardware limitations I could find on the rpi's bluetooth chipset.
Sorry if any of these questions seem obvious as I am quite new to IoT development and BLE as a whole.

Code in use:
```python
comms_device = peripheral.Peripheral(list(adapter.Adapter.available())[0].address,local_name="Device Name")
comms_device.advert.local_name = storage.Storage.device_code
comms_device.advert.manufacturer_data(company_id=0xFFFF, data=Ble.advert_data)
comms_device.publish()

# Called later within a function
comms_device.ad_manager.unregister_advertisement(cls.comms_device.advert, {}) # Doesn't work
````

Thank you in advance and any help would be appreciated! :)

ukBaz, 2024-10-04:
You are correct that a requirement of BlueZ is that you must first stop the advertisement and create a new one. You cannot modify and already running advertisement.

"Doesn't work" is not particularly useful at giving clues as to what is wrong. I'm not in a position to test this right now, but if I were to speculate, then `unregister_advertisement` only takes one argument which is the D-Bus of the advert. So I would expect your code to look like:
```python
comms_device.ad_manager.unregister_advertisement(cls.comms_device.advert.path)
```

AKFTHEGREAT, 2024-10-04:
Thank you very much for your reply on such short notice.
Your fix does indeed work, it was as simple as just removing the second argument. 
Thank you also for your work on this project as a whole, as it has been a phenomenal help on the project I am currently working on.
Have a great rest of your day!

closed 2024-10-04 by AKFTHEGREAT