ukBaz/python-bluezero #421: How to make a peripheral asynchronous

ikarus23, 2024-12-29:
Hi. Thanks for the great project!

I'm playing around on implementing something like a BLE keyboard. However, I have a hard time understanding how to make it asynchronous. Let's say I can setup all services and characteristics correctly. I can make an function that uses a timer from the `async_tools` to e.g. press the key "A" every 5 seconds. But I want to send an "A" (notify a subscribed central) whenever I want -- e.g. when the user enters an "A" I want to emulate typing that "A".

Can you point me in the right direction? Is that something for asyncio? Threading? Have I missed something? I'm not familiar with the GLib loop stuff in the library/the examples.

ukBaz, 2024-12-29:
This is quite a broad and generic question that has lots of detail behind it depending on exactly what you need so it is going to be difficult to answer it here with the information you've given. As some general pointers...

I have done a gist about [creating a Bluetooth Keyboard with a Raspberry PI](https://gist.github.com/ukBaz/a47e71e7b87fbc851b27cde7d1c0fcf0) that sounds similar to what you are looking for.

However, it emulates a Bluetooth Classic keyboard (HID) rather than the BLE equivalent [HID over GATT](https://www.bluetooth.com/specifications/specs/hid-over-gatt-profile-1-0/) which is sometimes known as HoG.

BlueZ (The Official Bluetooth stack on Linux) uses the main event loop from GLib and GTK for asynchronous sources. There are plenty of examples of connecting GLib and asyncio event loops (although I've always been able to manage with just the [GLib event loop](https://docs.gtk.org/glib/main-loop.html) for my applications although your mileage may vary

ikarus23, 2024-12-30:
Hi. Thanks for your answer. I remember my colleague also implementing/adopting a Bluetooth Classic keyboard (https://github.com/SySS-Research/bluetooth-keyboard-emulator). This works, but yes, I need BLE. Not even a keyboard. Just a BLE peripheral sending at random points in time (user controlled).

I was not aware of BlueZ using the GLib main event loop. Truth be, I did not even know about GLib and the event loop. That is probably the reason I had a hard time to understanding the asynchronous working of this project and BlueZ in general. I will look into it. I'm pretty sure that will clear things up for me.

Thanks.

closed 2024-12-30 by ikarus23