ukBaz/python-bluezero #423: Setting new set of advertisements produces NoReply error

andpuc23, 2025-01-23:
I use bluezero to advertise the data to BLE devices around me without pairing to them. 
I try to implement the following sequence: (it works in Java already, using that is not an option)

1. Advertise "query" command to devices
2. Each device responds with its unique id
3. Advertise "flash" command with specific device id (all devices whose id is different ignore the command)

I can send "query" and receive the responses (using [bleak](https://github.com/hbldh/bleak) library), the problem arises when I need to send 2-nd command.

Here's the code stripped of sensitive and non-relevant data:

```
class BluetoothSender:
    beacon = broadcaster.Beacon()
    def __init__(self):    
        self.ad_manager = advertisement.AdvertisingManager(self.beacon.dongle.address)
        self.ads = []
        
    def add_advertisement(self, manufacturer_data, uid):
        global ad_id
        new_ad = advertisement.Advertisement(ad_id+1, 'broadcast')
        ad_id += 1
        new_ad.manufacturer_data(INGY_ID, manufacturer_data)
        # converting uid to format "12345678-1234-1234-1234-1234567890AB"
        new_ad.service_UUIDs = [uid_str] # in 
        self.ads.append(new_ad)
        self.ad_manager.register_advertisement(new_ad)

    def start_advertising(self):
        if not self.beacon.dongle.powered:
            self.beacon.dongle.powered = True
        for ad in self.ads:
            ad.start()
        
    def unregister_ads(self):
        for ad in self.ads:
            ad.stop()
            self.ad_manager.unregister_advertisement(ad)
        self.ads = []

    async def advertise_for(self, timeout_seconds):
        p = Process(target=self.start_advertising)
        p.start()
        await asyncio.sleep(timeout_seconds)
        p.terminate()
```

driver code:
```
beacon = BluetoothSender()

beacon.add_advertisement(manufacturer_data, uuid)

task = asyncio.create_task(beacon.advertise_for(time))
await task

 uuid, man_data = flash_command.encode()

# tried with the old beacon object as well
del beacon
new_beacon = BluetoothSender()

new_beacon.add_advertisement(man_data, uuid)
new_task = asyncio.create_task(new_beacon.advertise_for(time))

await new_task
```
Everything goes fine until I try to advertise the 2-nd time. On ```ad.start()``` DBus throws an error:
```ERROR:dbus.proxies:Introspect error on :1.295:/org/bluez/hci0: dbus.exceptions.DBusException: org.freedesktop.DBus.Error.NoReply: Did not receive a reply. Possible causes include: the remote application did not send a reply, the message bus security policy blocked the reply, the reply timeout expired, or the network connection was broken.
Failed to register advertisement: org.freedesktop.DBus.Error.UnknownMethod: Method "RegisterAdvertisement" with signature "sa{sv}" on interface "org.bluez.LEAdvertisingManager1" doesn't exist

I found this [answer](https://stackoverflow.com/a/75099825) at stackoverflow, however I do not understand if this is done when I create a new beacon object (I guess so?) or not.

In any case, there is this issue and I require your help.

Thanks in advance!

ukBaz, 2025-01-24:
Something very similar to this was done in https://github.com/ukBaz/python-bluezero/issues/242 
Maybe take a look at that to see if it helps you. If not, please add more information here.

andpuc23, 2025-01-27:
ukBaz, thanks for your response! I have checked out #242 
and no, it doesn't work for me. Specifically:

BluetoothSender class:
```
    def unregister_ads(self):
        print('BluetoothSender: unregistering advertisements:', end='\t')
        self.beacon.broadcaster.mainloop.quit()
        for ad in self.ads:
            # ad.stop()
            self.ad_manager.unregister_advertisement(ad)
        self.ads = []
        print('Unregistered')
```

Updated driver code from OP:
```beacon = BluetoothSender()

beacon.add_advertisement(query1)
beacon.add_advertisement(query2)
scanner = BluetoothScanner()

scan_task = asyncio.create_task(scanner.scan_for(time*2))
advert_task = asyncio.create_task(beacon.advertise_for(time))
    
    
await asyncio.gather(
        advert_task,
        scan_task
    )
   
print('finished advertising')
beacon.unregister_ads()
print('unregistered ads')
```

This provides me with output:
```
BluetoothScanner: start scan
BluetoothSender: starting advertisement
BluetoothSender: ad [query1] registered
BluetoothSender: ad [query2] registered
BluetoothSender start_advertising: process terminated
BluetoothScanner: scanning finished
BluetoothScanner start_scan: process terminated
finished advertising
ERROR:dbus.proxies:Introspect error on :1.3:/org/bluez/hci0: dbus.exceptions.DBusException: org.freedesktop.DBus.Error.NoReply: Did not receive a reply. Possible causes include: the remote application did not send a reply, the message bus security policy blocked the reply, the reply timeout expired, or the network connection was broken.
BluetoothSender: unregistering advertisements:	Traceback (most recent call last):
  File "/home/andrei/code/robot/BLE/ble-python/ble_tests.py", line 84, in <module>
    asyncio.run(main(int(sys.argv[1])))
  File "/usr/lib/python3.10/asyncio/runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "/usr/lib/python3.10/asyncio/base_events.py", line 649, in run_until_complete
    return future.result()
  File "/home/andrei/code/robot/BLE/ble-python/ble_tests.py", line 49, in main
    beacon.unregister_ads()
  File "/home/andrei/code/robot/BLE/ble-python/bluetooth_utils.py", line 65, in unregister_ads
    self.ad_manager.unregister_advertisement(ad)
  File "/home/andrei/.venvs/macaquentosh/lib/python3.10/site-packages/bluezero/advertisement.py", line 333, in unregister_advertisement
    self.advert_mngr_methods.UnregisterAdvertisement(
  File "/home/andrei/.venvs/macaquentosh/lib/python3.10/site-packages/dbus/proxies.py", line 72, in __call__
    return self._proxy_method(*args, **keywords)
  File "/home/andrei/.venvs/macaquentosh/lib/python3.10/site-packages/dbus/proxies.py", line 141, in __call__
    return self._connection.call_blocking(self._named_service,
  File "/home/andrei/.venvs/macaquentosh/lib/python3.10/site-packages/dbus/connection.py", line 634, in call_blocking
    reply_message = self.send_message_with_reply_and_block(
dbus.exceptions.DBusException: org.freedesktop.DBus.Error.UnknownMethod: Method "UnregisterAdvertisement" with signature "s" on interface "org.bluez.LEAdvertisingManager1" doesn't exist
```

I have tried `unregister_advertisement()`, it didn't work for me so I tried to go without it. Didn't work as well. I am not quite familiar with dbus, so maybe it is something obvious I don't see.

ukBaz, 2025-01-27:
There is a lot to unpack in what you have done. I don't have time to debug this now, but here are some observations from just reading through your code extracts.

First thing to note is that you are using two event loops. The first is what BlueZ (and Bluezero) uses which is the GLib Main Loop. You are also using Python's asyncio to add a timeout. I've seen both used but it always fills me with fear as it is difficult and error prone. Is there a reason you can't use the GLib main loop? There are examples of how to do the timeouts in https://github.com/ukBaz/python-bluezero/blob/90f38a0fefcd3d4f4e00fbb4af76bd132a0e74f4/bluezero/async_tools.py#L14-L27

This is what is used in the example that I linked to in #242.

Secondly, in your code above it looks like you are stopping the GLib "mainloop" which I am assuming is why you are getting the `no reply` error when you try to unregister the advertisement; You have effectively stopped your app running so it doesn't respond.

In your code you appear to be creating two advertisements. From your description at the start it sounds more like you want scan response data. I don't have experience of working with scan response in BlueZ as it was (is?) hidden behind the experimental flag in the Linux distribution I use:
https://git.kernel.org/pub/scm/bluetooth/bluez.git/tree/doc/org.bluez.LEAdvertisement.rst?h=5.72#n86

You may find the following article helpful in explaining the different parts of advertisements:
https://jimmywongiot.com/2019/08/13/advertising-payload-format-on-ble/

I would suggest you strip your code back to a Minimal Runnable Example if you want me to help with the debug.
Maybe start with just a static advertisement that doesn't change. Once we are confident that the format of the advertisement is correct, then we can move on to having it dynamically change.

It would also be helpful if you could use the [nRF Connect](https://www.nordicsemi.com/Products/Development-tools/nRF-Connect-for-mobile) app to check the advertisement is in the correct format. Then we can both use the same tool.

andpuc23, 2025-01-28:
ukBaz, thanks a lot for the suggestion with GLib's event loop. Tried it out, sending several advertisements works amazing!

What I did:
```
class BluetoothSender:
    
    beacon = broadcaster.Beacon()
    def __init__(self):    
        self.ad_manager = advertisement.AdvertisingManager(self.beacon.dongle.address)
        self.ads = []
        
    def add_advertisement(self, command=None, uid=None, man_data=None):
            new_ad = advertisement.Advertisement(some_random_number, 'broadcast')

        new_ad.manufacturer_data(data)
        new_ad.service_UUIDs = [another_data]

        self.ads.append(new_ad)

        def error_handler(error):
            print('BluetoothSender: error while registering advertisement')
            print(error)
        self.ad_manager.advert_mngr_methods.RegisterAdvertisement(new_ad.path,
                                                                      dbus.Dictionary(dbus.Array(), signature='sv'),
                                                                      reply_handler=lambda: print(f'BluetoothSender: ad registered'),
                                                                      error_handler=error_handler)

    def start_advertising(self):
        if not self.beacon.dongle.powered:
            self.beacon.dongle.powered = True
        print('BluetoothSender: starting advertisement')
        for ad in self.ads:
            try:
                ad.start()
            except Exception as e:
                print('got error', e)
        
    
    def unregister_ads(self):
        print('BluetoothSender: unregistering advertisements:', end='\t')
        for ad in self.ads:
            ad.stop()
            self.ad_manager.unregister_advertisement(ad)
        self.ads = []
        print('Unregistered')

def main(time:int):
    beacon = BluetoothSender()
    query = Command()
    beacon.add_advertisement(query)    

    add_timer_seconds(5, beacon.unregister_ads)
    beacon.start_advertising()

    print('finished advertising')
    beacon.unregister_ads()

    if not scanner.devices:
        print('No devices detected, exiting')
        # return
    
    flash = Command()

    beacon.add_advertisement(flash)
    add_timer_seconds(5, beacon.unregister_ads)
    beacon.start_advertising()

    print('finished')
```

Not quite sure if I should `add_timer()` first and then `start_advertising()` or vice versa, but anyway this works.
Thanks one again!

closed 2025-01-28 by andpuc23