ukBaz/python-bluezero #426: How to get the written Data?

Thxlx, 2025-05-13:
Hello! 

I am currently trying to get bluezero to work on a Pi Zero W. Connecting to the server is working as expected and it prints the MAC of the connected device on connection and disconnection. However when trying to write with nRF Connect it doesn't print anything else. 

So I cannot get it to output more than this: 

```
Starting BLE GATT server...
Advertising as: RPiHeartRate
Advertisement registered
Connected to 5F:B6:xx:xx:xx:xx
Disconnected from 5F:B6:xx:xx:xx:xx
```

My Code:
```
from bluezero import peripheral

def write_callback(value):
        print("write works!")
        print(f"Received data: '{value}'")

def onconnect(ble_device):
    print("Connected to " + str(ble_device.address))
def ondisconnect(ble_device):
    print("Disconnected from " + str(ble_device.address))

periph = peripheral.Peripheral(adapter_address='B8:27:EB:CC:A4:43',
                                local_name='RPiHeartRate')

periph.add_service(srv_id=1,
                   uuid='12345678-1234-5678-1234-56789abcdef1',
                   primary=True)

periph.add_characteristic(srv_id=1,
                          chr_id=1,
                          uuid='12345678-1234-5678-1234-56789abcdef0',
                          value=[],
                          notifying=False,
                          flags=['write','write-without-response'],
                          write_callback=write_callback)

periph.on_connect = onconnect
periph.on_disconnect = ondisconnect

print("Starting BLE GATT server...")
print(f"Advertising as: {periph.local_name}")
periph.publish()
```

Thxlx, 2025-05-13:
Fixed it myself by using your example code and altering it. Sorry. :D 
Thank you for the great work!

niffler92, 2025-06-01:
@Thxlx what was your fix? I'm struggling to get it working on Pi Zero W. It doesn't seem to register any service or characteristics.
$ busctl tree org.bluez
└─/org
  └─/org/bluez
    ├─/org/bluez/hci0
    └─/org/bluez/test