ukBaz/python-bluezero #427: How to listen for AVRCP cmds/events from Bluetooth speaker like play,pause,etc?

thePromger, 2025-09-27:
Use case:
I have pi zero, acting as a music player and Bluetooth speaker as a output device,
When I press play, paise, volume up or down keys then I need to listen for these events in python, so I can do perform actions accordingly

I am not able to receive any events in python, took some help of AI too  bluez directly
I even tried mpris it worked in desktop Linux but in pi os lite non-gui
I am not able to receive any events from Bluetoothd, 
I tried playerctl for testing, it was able to send events to my mpris python player but not directly from Bluetoothd

I even tried mpris-proxy in pi os lite but that even didn't worked

Bluez docs and api are so confusing
MediaControl1, MediaPlayer1, Media1 which one is for what?

Can you do please help like what am I doing wrong?

WayneKeenan, 2025-09-27:
The connection might be using Classic Bluetooth therefore BlueZero can't help as I only supports  Bluetooth LE.

Have a look at the test script at the end of this: https://chatgpt.com/share/68d76c86-da14-8005-9d33-904c96c6096d

thePromger, 2025-09-28:
@WayneKeenan 
Do you know any repo for classic bluetooth in python for Linux?

I tried bluez method by gpt, but didn't worked at all
I don't receive any event in it

WayneKeenan, 2025-09-28:
@thePromger yes, the "Bluetooth Comm API" that comes as part of Blue Dot: https://bluedot.readthedocs.io/en/latest/btcommapi.html

thePromger, 2025-09-28:
> @thePromger yes, the "Bluetooth Comm API" that comes as part of Blue Dot: https://bluedot.readthedocs.io/en/latest/btcommapi.html

Thanks it a good project
But it's more like control pi using Bluetooth

My use case is simple, I am gonna use my custom music player in python, when I press play/pause btn on bluetooth speaker, headset
Player should receive it and then play the song, same way how ble speaker works with phones

ukBaz, 2025-09-28:
Hi @thePromger, I think I see the crux of what you're asking—and why the responses so far might be missing the mark.

You're not trying to expose a media player to a remote device (as Bluezero’s MediaPlayer1 interface does), but rather trying to listen for AVRCP commands (like play/pause/volume) coming from a Bluetooth speaker that’s paired to your Pi Zero.

That’s a subtle but important distinction. Most of BlueZ’s D-Bus interfaces—MediaPlayer1, MediaControl1, etc.—are geared toward controlling remote players, not receiving button events from speakers.

From what I understand, AVRCP button presses from a Bluetooth speaker aren’t exposed via BlueZ’s D-Bus API by default. On desktop Linux, MPRIS and playerctl can sometimes bridge that gap, but on Pi OS Lite (without a full desktop stack), those layers aren’t present.

So far, I haven’t seen a clean way to capture AVRCP events like volume or play/pause from a speaker using BlueZ alone. You might need to:
- Use btmon or dbus-monitor to see if anything is emitted at a lower level.
- Explore whether bluetoothd can be configured or patched to expose those events.
- Or consider a custom BlueZ plugin (though that’s a deep dive).

If anyone has managed to capture AVRCP events from a speaker on Pi OS Lite, I’d love to hear how they did it. But I suspect this is a gap in what BlueZ exposes by default.

Hope that helps clarify the distinction—and why the usual media player interfaces might not be the answer here.

As this doesn’t appear to be a direct issue with the Bluezero library implementation, I’m going to close this for now. But if I’ve misunderstood your intent or if there’s something Bluezero could do to help surface these events, please feel free to re-open or continue the conversation.

closed 2025-09-28 by ukBaz

thePromger, 2025-09-29:
@ukBaz 

Yep, you're right I didn't understood bluez api correctly

I just stepped into bluez world a week ago
So, its taking sometime to understand the docs, api, all these things

I yesterday tested media Player in Linux acted as speaker and phone acting as source/player to play audio
To check how MediaPlayer bluez api works and it worked
But my use case is vice versa, where Linux will act like Source

+ I go through bluez code too, where I find that they send avrcp events directly to uinput, which can be used if MediaPlayer api doesn't worked for me at all

+ Mpris player method also their but it won't work in non-gui os. So not for me

+ In bluez api docs, they have an api to register the media player, so when event occurs bluez can call those methods like what action to perform on play, pause, next, etc
They don't provide any method or api to listen for those events directly except if you run bluetoothd in debug mode

I think this once works that's why they have gave it in docs, but I am not able to register the player correctly that's why I am not able to see it in logs

+ In bluetoothd debug mode , I am able to see the avrcp commands but not in dbus-monitor when Linux is a player/source, else it prints some thing on the logs

I am trying to register the media Player in Bluez MediaPlayer api
Maybe I am not able to get it correctly

Can you help me in this case, codes I find are old which aren't working correctly at all, tested so many times

So can you help me with codes to how to register the media player on MediaPlayer1 interface using dbus for bluez, so it can call those methods when avrcp events received

+ 
I came to your repo for help after reading your bluez article
So maybe it's irrelevant to this repo but I didn't find any person or platform where I can ask and get the answer too

Note: Anyone came here in future, to read about how to use bluez APIs for avrcp events, then do try to use mpris first, it has good api/support, easy to use + extraMethod also there for an application needs